AIRLegend/aitrack #117: I have found the source of a major bug

AMKaxel, 2021-09-14:
I have found the source of a big bug. Its the log. 
Sometimes when installing the program it simply just doesn't run. This is caused by the Avast Anti-Ransomware module. 
When you install the program in a protected directory and you run it for the first time the program automatically creates a log file. This is what triggers the Anti-Ransomware module. It detects an unknown program crating/modifying files, and it kills the process. 

A way to solve this problem is marking the program as non malicious in the Anti-Virus. 

Or the program comes already installed with the log file. (Not sure if this could solve it). 

Ps. I know it can be frightening marking a program that you don't know as non malicious, especially when your Anti-Virus blocks it, but trust me, I have scanned this program with over a hundred Anti-Virus software's, and no one has found something sus. Sometimes Anti-Virus just over react.

AIRLegend, 2021-09-24:
Yeah, that must be certainly annoying. I'll change the title to  make it clearer.

If any dev knows how to "sign" (or whatever) the executable to be more Antivirus-friendly, a PR will be welcome :)

searching46dof, 2022-06-25:
You can sign for free using https://letsencrypt.org/

However, it will not make it more antivirus friendly. Signing using a 3rd party certificate authority only authenticates the file came from the certificate owner and allows the file to be distributed by 3rd parties where the end user can detect if the file has been modified.

You can use https://www.virustotal.com/gui/home/upload to scan the file with multiple engines. If its only detected by a single engine (of a few since there may be more than one product using the same engine) and the file is older, its more likely to be a false positive detection. If its a very recent file, it may either be a false positive or a zero day (new) virus.

From the description the Avast Anti-Ransomware module is using a heuristic detection based on behavior. Heuristic detection is very prone to false alarms.

The sample program should be submitted to Avast so that it can be excluded by its private internal exclusion list