AIRLegend/aitrack #118: 0.64 and 0.65 losing focus

winterwicht, 2021-09-17:
A bit head movement left or right and the face focus is lost, 0.63 is better

Steps to reproduce the behavior:
Install 0.64 and/or 0.65, move,turn head left or right ;)

**Environment (please complete the following information):**
Win 10
 - Webcam model: 
 - Logitech C270
 - AITrack version:
 - 0.64 and 0.65

rippinggail, 2022-02-20:
Using same webcam, also Windows 10; found the same performance. Tracking in release .65 is much worse for me than in .63. 
In .65, aitrack can't find my face if I turn more than maybe 10 degrees. .63 seems much more reliable for me.

Rizz22, 2022-06-25:
Having same issue on Windows 10. I use a cheap webcam though.  0.63 able to track my face reliably but 0.64 and 0.65 keeps losing focus. Especially if my face gets closer, 0.63 still detects my face while 0.64 and 0.65 just don't.