AIRLegend/aitrack #120: Could the software detect two faces?

frankzhang196, 2021-10-01:
**Describe your idea**
A clear and concise description of what the idea. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

AIRLegend, 2021-10-01:
Why would that be needed? I mean, I suppose you're talking about having two tracking data streams. However, I  don't see how that could be compatible with opentrack as it only allows a single data stream.

frankzhang196, 2021-10-02:
> Why would that be needed? I mean, I suppose you're talking about having two tracking data streams. However, I don't see how that could be compatible with opentrack as it only allows a single data stream.
Thanks for your reply! You mean, enve if I change the source code to detect 2 faces,  will not use opentrack to tracking 2 faces. Its helpful, thank you. I am interested in the models for detecting face and landmarks, could you share related papers or docs about the architecture of the model?

searching46dof, 2022-08-31:
To frankzhang196,
You may want to try version 0.6.6alpha which uses cv:FaceDetectorYN and detects multiple faces.
In addition to modifying aitrack to output multiple data streams, you would need to add some configuration/intelligence to associate a face to a specific data stream.

frankzhang196, 2022-08-31:
> To frankzhang196, You may want to try version 0.6.6alpha which uses cv:FaceDetectorYN and detects multiple faces. In addition to modifying aitrack to output multiple data streams, you would need to add some configuration/intelligence to associate a face to a specific data stream.
> 
> I submitted changes for adding center weighting (post version 0.6.6) by detecting the face closest to the center of the preview window. It could be enhanced to detect the face closest to multiple points in the preview window.

thanks!

DrJaymz, 2022-12-05:
I would like a setting that if it sees two it just returns centre position.  Because if I'm on the flight simulator and one of my kids comes in it goes completely nuts.

searching46dof, 2022-12-06:
@DrJaymz , version 0.7.0 already implemented center weighted face tracking

DrJaymz, 2022-12-06:
I haven't had the kids come in with the latest version.  The kids just think I sit there crashing the plane all night.