AIRLegend/aitrack #132: Face not detected at small pitch down angles

mahela007, 2022-03-09:
**Describe the bug**
My face stops being recognised at very small pitch down angles..

**To Reproduce**
Start tracker
Landmark stabilization is on.

**Expected behavior**
My face should be tracked at least until about 45 degrees downward. But it stops after about 25 degrees downward rotation.

**Screenshots**

**Environment (please complete the following information):**
 - Operating system version (Windows 10+ Version number): 
 - Webcam model: droidDam Motorola Moto G
 - AITrack version: 6.1

It seems to depend on camera position..my camera is mounted above my monitor.

LumberingTroll, 2022-03-22:
I get this as well, what I do to compensate is increase the curve to be much steeper when looking down, with a slightly larger dead zone. that way I don't have to look down nearly as far to get the full angle. 

Better tracking solution would be wonderful though.

Virtike, 2022-05-19:
I have found that having the webcam mounted under the monitor rather than above helps a lot with this. Seems to track better looking slightly up at a face rather than down.

Lothdeorn, 2022-07-19:
I less FOV value blank since i don't know the value of my webcam. (if you don't know your fov, don't put a random value)
I augmente the exposure with the software of my webcam to get a better contrast.
I put the model type in heavy to allow the app to make the pink dot at the good place.

It's not perfect but it's the best setting i could get to get over this issue.

searching46dof, 2022-07-27:
The pitch detection appears to be very sensitive to the ambient lighting. I find that it is optimal with direct light just above the front of your face to promote shadows with the camera below the center of the monitor.  It can detect +/-30degrees. Reflected, diffuse, or ambient light makes the detection less sensitive to approx +/-20degrees.

You will need to set an accurate FOV otherwise it will make both the pitch and yaw either less or too sensitive. The project's wiki has FOV for some cameras. If you cannot obtain the information from the manufacturer's website, you can always measure the approximate angle between opposite corners of the preview window from the camera's perspective.  In version 0.6.5a, the sources indicate that you need to explicitly the FOV and your face should be as close to the center of the preview window.

The insensitivity due to lighting appears to be function of determining the aspect ratio of the mouth from shadows. A simple workaround is to change the shape of the lips when this occurs. You can try either an exaggerated pouting/puckering or an exaggerated smile to add or subtract pitch.