AIRLegend/aitrack #134: No video input

BlueStrikePro, 2022-04-09:
So I wanted to make eye-tracking through my phone to play euro truck simulator 2. I followed a tutorial by 
A. P. 101 Gaming

The problem is, is that it says No video input, and when I click Start tracking, All I see is a black screen and a glitch line every 1 second

[image]

[image]

OS Windows 11
Camera: Droidcam source 2 or 3 Samsung Galaxy A6      USB and Wifi
AI Track version v0.6.4

d-kleine, 2022-05-08:
Have you tried using a different camera from the drop-menu in the left upper corner?
Usually, the camera being used as a webcam is 0 (-> "Camera 0")

Flexxkii, 2022-07-06:
Unfortunately, I got the same issue :/