AIRLegend/aitrack #135: Exe not included in release zipped file

mangel890, 2022-04-30:
Hi,
I might be wrong but I've been trying to find AItrack.exe file as indicated by the instructions and I cannot find it anywhere. In releases page (https://github.com/AIRLegend/aitrack/releases), I open the latest available ([aitrack-v0.6.5-alpha.zip](https://github.com/AIRLegend/aitrack/releases/download/v0.6.5-alpha/aitrack-v0.6.5-alpha.zip)) and I can see the content of the github repository but there is no executable binary. 

Is it missing for any reason? If so, is there a plan to get it back there?

I have also checked the build page from appveyor but it seems binaries are deleted after some time, so it's not present anymore.

It seems the only alternative is to build it ourselves locally?

Thank you very much,
BR.

AIRLegend, 2022-04-30:
Hello,
Just checked and it's there: 

[image]

Maybe your AV is automatically deleting it (?)

mangel890, 2022-04-30:
Thanks for your quick reply. Indeed the file is there. The firewall is blocking the download because it finds suspicious to have an exe inside a zip.
On the other hand, the other download (src code) has a similar name (only difference is a 'v' before the version) and it got downloaded correctly. Due to the similarity of names I thought it was the release zip and I got confused looking for the exe in the src code :)

Thanks again for your quick reply.
BR.

closed 2022-04-30 by mangel890