AIRLegend/aitrack #137: Camera Working but no tracking. No pink dots/blue box

ColoradoBMR, 2022-05-22:
**Describe the bug**
Everything seems to work fine except the actual tracking. I get a preview, no problem but no pink dots/blue box. I am using an older Logitech but it meets the min req. I've tried messing with the resolution to no avail. I've moved the distance back and forth, and played around with FOV

Any other suggestions? I haven't found this issue or a resolution after a good hour or searching.

searching46dof, 2022-06-25:
I'm using 0.6.5 and found that detection works best at approximately 1meter where the preview image is like a portrait painting.
Camera position depends a lot on the lighting.

In one room where there is a torch lamp to one side of the tv, the light is reflected from the ceiling. If I place the camera on top of the tv, it cannot detect my face. But if I place it below the tv, it can detect my face but can only detect a pitch up of 20 degrees where there is no problem with pitch down. It behaves the same either a distance of 4 meters away on a sofa or standing up close less than a meter. There does not seem to be any issue with yaw range even though the light is coming from a corner of the room

In another room where there is a center ceiling light, the light is direct. I can place the camera on top or below the tv. There is no problem with the pitch range up close or 4 meters away.

Rizz22, 2022-06-25:
I just put comment to another issue #118. Seems to be same problem.
0.64 and 0.65 unable to detects my face when I get closer to the cam but 0.63 working perfectly.

Around 70 cm from the camera it able to detects (haven't try some extreme degrees though) and seems working well.
Sometimes I need to zoom in by get closer to the screen (to read some text).  At around 50 cm it just loses focus instantly.

Tweaked the distance and FOV settings but seems no effect.

searching46dof, 2022-06-26:
i previously was using opentrack's accela filter with aitrack 0.65 gain/exposure approx 60% in the evening with house lighting.
i had problems where selecting other opentrack filters would prevent aitrack face tracking.

I decreased the gain/exposure to approx 40% for brighter natural lighting in the early afternoon without any house lighting. selecting other filters no longer prevents face tracking.

searching46dof, 2022-08-31:
With version 0.6.5 there was an issue tracking at very close distances (less than 1 meter) and with less than optimal lighting (directly overhead).
Also manually adjusting the gain/exposure provided much better results than auto. You need to roughly white balance the preview window which will depend on the camera model and ambient lighting.