AIRLegend/aitrack #142: Problem Initializing the tracker

OleRodenburg, 2022-06-11:
**Describe the bug**
When opening AITrack.exe from version 0.6.5 i get the popup: 'There was a problem initializing the tracker. Check the models folder and restart the program.' 

The program opens when i click OK on the pop up but i can't do anything with it afterwards.

AIRLegend, 2022-06-12:
Hi! 
Make sure the exe is inside the downloaded folder (next to the other files) and that you have read/write permissions.

searching46dof, 2022-10-08:
Are you able to use your webcam with another application?
I get this symptom when there are no webcams connected.

If this is a PS3Eye, you may want to try installing the driver at https://github.com/jkevin/PS3EyeDirectShow
I believe there may be more than one hardware revision for the PS3Eye