AIRLegend/aitrack #143: Crashes When pressing 'Start Track'

ronaldoks7, 2022-06-13:
When I launch AiTrack, nothing happens when I click it/open with Windows 10 menu (bottom left corner). I need to run it as administrator, which then launches the app. However, when I click start tracking, it either tracks for 5 mins then the camera goes ballistic, and goes black with random flickers of colour, or just crashes and closes automatically.

1. Run AiTrack as administrator
2. Click on 'Start Tracking'
3. App freezes and crashes, or camera goes black in the 'Enable Preview' box. Or random flickers of coloured lines and an occasional flash of my face.

Expected Behaviour:
1. Run AiTrack as administrator
2. Camera turns on and records smoothly in real time in the 'Enable Preview' box.
3. Head is placed properly and aligned in the cockpit.

No screenshots as it mostly just crashes.

Environment:
- Windows 10 Pro (most recent version)
- Nulaxy model C900
- AiTrack version 0.6.5

AIRLegend, 2022-08-14:
Hey :)  That sounds like the resolution/fps parameters you set are not compatible with your particular camera model.

searching46dof, 2022-08-31:
This may be a driver issue since it tracks for approx 5 mins before it goes ballistic.
Check if there is an updated driver from the manufacturer or if there is a WHQL driver from Microsoft.
You may also want to verify if the driver/camera is stable with another app.

Also check that the chipset drivers are up to date since it could cause that type of symptom.

Another possibility is that the camera is underpowered. Try connecting the camera to the pc via a powered usb hub.