AIRLegend/aitrack #147: High CPU usage

dv1rb, 2022-06-27:
Hello, I have a bug with Ai track where when I "start tracking"  it uses a lot from my CPU (10%- 20%+). I have tried lowering the resolution and fps but it stays the same. I would apricate any help.

[image]
[image]

 - Operating system version (Windows 10 Version number): 21H2
 - Webcam model: silver line (i dont remember the exact model but it's a new full HD model)
 - AITrack version: v0.6.5

searching46dof, 2022-06-29:
you can try setting a custom brighness and exposure. you will need to enable the preview pane to determine the exposure level. if it's too dark or too light it will be more difficult to detect the face and use more CPU. I am using both a PS3 Eye Cam and a Rosewill RCM-8163. All other things being identical, I find the selecting the PS3 Eye Cam uses less cpu than the Rosewill, probably due to the drivers. Those settings may also depend on amount of lighting during a time of day. I set the gain/exposure to slightly overexposed (cool vs warm picture) which seems to detect a little easier.

you can also try selecting the faster model type to reduce CPU. I'm using Heavy to try to reduce jitter.

You can also try unchecking autocheck updates since it's not necessary for normal operation.

I also use remote opentrack client set for 127.0.0.1 (localhost) with the same port that opentrack udp input client is configured for.

If this is a recent webcam, especially HD, verify the Field Of View is correct. 56deg seems very narrow since most webcams now are usually 70-110deg. I bought a PS3 Eye Cam since you can rotate the lens between telephoto (56deg) for a narrow FOV and wide (70deg) in order to reduce the amount of background.

Also try varying the distance to the camera.  At a close distance such as 0.7m your face may be filling up the entire video window and there will be more pixels to process for face detection. I had best results between 1 and 1.5meters but am currently using it at 4meters from a tv.

What CPU/speed are you using? I'm using a Ryzen 5700g (default speed) where 30FPS is ~6-7% and 60FPS is ~10-12% so I use it at 30FPS

dv1rb, 2022-06-29:
Hello, thanks for responding. I have tried changing everything in the settings but nothing lowers the CPU usage. my CPU is an i7-7700. changing the fps, fov, resolution, distance and model type didn't change the CPU usage it's till at 20%.

searching46dof, 2022-06-30:
I tried a using a webcam on a i7-2600. Initially, I could only get a black preview screen and it took 31% cpu.

I then used a powered usb2.0 hub between the computer and the webcam. Then I am getting the same results as on a Ryzen 5700g. Approx 7% CPU at 30FPS and 13% at 60FPS. Setting the model type between Fast and Heavy did not seem to make a difference.

The webcam was probably underpowered and it's response was slower when connected directly to the computer's usb 2.0 port.

searching46dof, 2022-06-30:
if your computer has both usb2.0 and usb3.0 ports, try connecting it directly to a usb3.0 port which can provide more power to the camera. it may be trying to run in a low power mode.

dv1rb, 2022-06-30:
I tried all the usb ports on my computer but it didn't change anything it's still above 15%.

searching46dof, 2022-07-01:
do you have a powered usb hub? if not, you may want to borrow one for a test.
you can also check for usb performance/power saving configuration in the bios and device manager.

you may also want to update any chipset and usb drivers from the motherboard manufacturer, intel, and the usb manufacturer.

another possibilty is that your antivirus/antimalware is causing the high cpu. you can try adding aitrack as a exclusion or temporarily disable any realtime scans.

if you are overclocking, try running without any overclocking settings.
have you verified that the cpu, memory and bus (a.k.a FSB) are running at stock speeds?
have you verfied that the camera is detected and running at least usb2.0 speeds?

AIRLegend, 2022-08-14:
Hi @dv1rb. As a note, "Heavy" is the most CPU-intensive model. If this is impacting your performance on games I would suggest setting it to "Medium" (or "Fast" if that works well for you) and lowering the resolution (in case your camera allows it).

Apart from what @searching46dof has suggested, maybe you should give the latest version ([0.6.6](https://github.com/AIRLegend/aitrack/releases/tag/v0.6.6-alpha)) a try, which should have a lower CPU consumption.

I'm closing this issue for the moment as it isn't really a bug. In case this doesn't improve by upgrading, please, open a new one.

Have a nice day :)

closed 2022-08-14 by AIRLegend