AIRLegend/aitrack #159: v6,6 Not working in windows 10

BradleyGother, 2022-08-17:
The program launches v6,6, but is not starting once i press start to track in windows 10. It just closes the application. Ive tried to run as administrator as well, but it does the same thing.

It has IP address box selected as default & cannot start using USB connection.

* 6.5 works, but high CPU usage compared to 6.3 which I use now.

AIRLegend, 2022-08-17:
Are your camera settings right? What's the content of your `log.txt` file?

BradleyGother, 2022-08-17:
Hi AIRLegend/aitrack

I managed to get it working. I use a 4G phone as a camera.
It only works on 640x480 resolution - I have to set the droidcam app to use
480p.
I cant get it to work on 720p settings in droidcam & config for AITrack. I
tried 940x720 as per previous versions that worked, but it doesn't work.
Not sure how much the resolution affects the stability of the head tracking
if higher is better?

Thanks for this software, reply & help.

Regards
Brad

AIRLegend, 2022-08-18:
Glad it's working now :)

The settings you set in AITrack must be supported by your camera (in this case, DroidCam). If not, crashing is normal. 

As for the resolution, it shouldn't matter as, internally, all images are downscaled to 224x224.

Closing the issue then. Have a nice day.

closed 2022-08-18 by AIRLegend