AIRLegend/aitrack #160: Rotation values

jonipokka, 2022-08-17:
**Describe the bug**
There seems to be a problem with roll values. I read from the patch notes that the values are capped at +-90 but the issue is that the default value that opentrack picks up is 90 and roll only works in one direction (changing offsets etc in opentrack do nothing).

Since the default value that aitrack sends is 90 when my head is straight and the values are cappes at +-90 the tilt only works one way. When I tilt my head right roll works the value decreases 87....86....83... Etc in opentrack but when I tilt my head left the value is capped at 90 and nothing happens.

**To Reproduce**
Steps to reproduce the behavior:
Start the program and follow instructions to use aitrack with opentrack. Default settings in opentrack.

**Expected behavior**
As my head seems to be recognized correctly since everything else works wonderfully (yaw, pitch, xyz) I expect the roll feature to work aswell in both directions and not only when rolling to one side.

**Environment (please complete the following information):**
 - Operating system version (Windows 10+ Version number):  Windows 10
 - Webcam model:  Logitech C920
 - AITrack version: 0.6.6

searching46dof, 2022-08-20:
I cofirmed the behavior.

Roll should be limited between 0.0 and 180.0 and v0.6.6 is coded as:
    // Limit roll between -90.0 and +90.0 
    if (face_data.rotation[2] >= 90.0)
        face_data.rotation[2] = 90.0;
    else if (face_data.rotation[2] <= -90.0)
        face_data.rotation[2] = -90.0;

I submitted the changes to restrict only yaw and pitch when they were initially exceeding +/-90deg which did not make sense since my face would be looking away from the camera. Roll did not have this issue and was not a problem since it would lose face tracking at +/-45deg from center (90deg).

I would recommend adding code to show the restrictions but leaving them #ifdef'd out for performance. It can be re-enabled if needed.
#if 0
    // Limit roll between -90.0 and +90.0 from center (=90deg)
    if (face_data.rotation[2] > 180.0)
        face_data.rotation[2] = 180.0;
    else if (face_data.rotation[2] < 0.0)
        face_data.rotation[2] = 0.0;
#endif

AIRLegend, 2022-08-21:
Yup. The problem is with the range. 

I think it's better to simply fix the range (set the clipping for roll to 0-180) instead of trusting the face detector will lose track at 45º as we could change the detectors in the future.

@searching46dof's proposal is the fix. I'll change it as soon as I can.

AIRLegend, 2022-08-23:
Fixed on #163

closed 2022-08-23 by AIRLegend