AIRLegend/aitrack #168: OBS Virtual Cam not detected as source

srjwebster, 2022-09-08:
Hi there, in order to continue to use my webcam for other purposes than facetracking, it'd be useful to be able to use a virtual camera as input.

For instance, uncontroversial game Star Citizen wants to use the webcam simultaneously to do FOIP. 

I've attempted several times to get AITrack to use my virtual webcam output, it is working with other applications.

AIRLegend, 2022-09-08:
Hey!
Seems strange, I know other people used several virtual-cam programs (OBS, Manycam, DroidCam, ...) without issues. Maybe it's a config problem for you? As long as the camera is exposed as a windows media foundation device it should be detected 🤔

srjwebster, 2022-09-09:
Other softwares in this space do see the camera as an input, but when selected they show as a black screen or crash the application. I am on Windows 11 and the virtual camera doesn't show up as a device in the windows camera settings, so I guess that's why! It does work for teams, discord, etc.. but I'll look on what it'll take to get it into the Windows Media device list like the actual base webcam C920.

srjwebster, 2022-09-09:
I have rolled back to OBS 27.2.4 and used the virtual camera plugin (which isn't compatible with 28.1.0)

This worked perfectly first time, just leaving this here for anyone in the future who might experience the same thing

closed 2022-09-09 by srjwebster

arvabalazs, 2022-11-25:
Hi

I can report this is still an issue with OBS Studio 28.1.2 and AITrack 0.7.1-alpha.
I can select a camera (which I think is the virtual camera) and the preview is black with some small colorfull tearings. I can confirm that the virtual camera working in Zoom.

Any idea expect reverting the OBS version?

BR,
Balázs