AIRLegend/aitrack #171: AiTrack not responding when opened, tried reconnecting my camera etc didnt work (Razer Kiyo)

Inhaltsangabe, 2022-09-10:
[image]
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Operating system version (Windows 10+ Version number): 
 - Webcam model: 
 - AITrack version:

**Additional context**
Add any other context about the problem here.

searching46dof, 2022-09-25:
Does the camera work with another application?

You may want to verify that all files have been extracted from the archive into a folder where the login user has full rights. Also try login and run as admin.
Verify that the bios, chipset drivers, usb drivers and camera drivers are up to date.
Check that the application is not being blocked by an antivirus application.
And also verify the camera has sufficient power. Try connecting it via a powered usb hub.

Inhaltsangabe, 2022-10-03:
> Does the camera work with another application?
>
> You may want to verify that all files have been extracted from the archive into a folder where the login user has full rights. Also try login and run [...]

Yes 
Yes I tried that 
Yes 
No its not
It does

Could there be an issue with Windows 11? Idk much about programming so yeah
Thanks for the great work appreciate it!

searching46dof, 2022-10-08:
This does seem like an issue with Windows 11.

Verify that the application is not blocked by Windows Firewall.

Is AiTrack enabled under settings/privacy/camera?
If it is already enabled, try disabling it, reboot and then re-enabling it and reboot.

searching46dof, 2022-11-09:
Do you also have the same camera, Razer Kiyo?
Does the camera work with another camera app? If it works then it most likely rules out a driver issue.
Is there another application that is obtaining control of the camera?
Also verify that the device is enabled in device manager.
Check if there is a BIOS/UEFI setting which may be disabling the camera. Maybe USB related.

This link (https://support.microsoft.com/en-us/windows/camera-doesn-t-work-in-windows-32adb016-b29c-a928-0073-53d31da0dad5) suggests that a camera may be turned on/off by a hotkey

Also check https://support.microsoft.com/en-us/windows/manage-app-permissions-for-your-camera-in-windows-87ebc757-1f87-7bbf-84b5-0686afb6ca6b#WindowsVersion=Windows_10 and https://support.microsoft.com/en-us/windows/windows-camera-microphone-and-privacy-a83257bc-e990-d54a-d212-b5e41beba857

I've had Win10 cases where the camera wasn't working with correct privacy settings and toggling the privacy settings off/on re-enabled it.

bbecausereasonss, 2022-11-09:
Mine has been crashing in Win10 after I click calibrate or start tracking. No log error.

searching46dof, 2022-11-09:
If this is an external usb camera, also try connecting it via a powered usb hub preferrably to a usb3 or usb2 port. I've experienced symptoms on one pc where it crashes on start tracking but also running at high cpu/slow FPS on another pc due to this.

TheExium, 2022-11-09:
Not the same camera for me. 
Yesterday morning yes, yesterday night no… In the device manager, it appeared as a webcam yesterday morning... but not last night... it only appeared as an audio source. I tested on a hub, it does not change anything.

my camera is recognized as usb audio, impossible to change the .sys file for usb video

searching46dof, 2022-11-10:
@TheExium,
what model camera, and motherboard are you using ?
are you able to use the camera with another application?

if you connected it via a powered usb hub and it made no difference, then it's not a power issue.
does the same symptom persist if you connect it to different usb ports on the computer?
can you verify if they are usb3/usb2 or usb1?

if it sometimes shows up only as a usb audio device, something is interfering with the video camera driver.
check if the same problem exists in an earlier driver and also check for a more recent driver.
Also try testing with an antivirus exclusion for Aitrack. You may also want to scan your PC for malware.
And also try testing with all overclocking disabled (CPU, CPU bus, memory)
And also try disabling any power saving mode for the webcam or usb port its attached to.

Are there other abnormal symptoms on your pc since the webcam driver seems unstable?
I would recommend running some diagnostics. If it is a notebook, it may already have some diagnostics in the BIOS/UEFI.
Otherwise, you can try prime95 and memtest86 to check the stability of the CPU, memory and CPU bus.

I would also recommend updating the USB drivers to the manufacturer specific drivers to support the specific USB chipset.

TheExium, 2022-11-10:
@searching46dof 
thank you for your help.

It’s a T’nB 1080, it’s a plug n play webcam. I have no driver, windows install it alone like an usb key.
I tried to update with the device manager… nothing… and i can’t try an older driver.
My MB is Asus rog strix b550 f gaming.

My computer is not OC, avast is ok with the permission of AITRACK, i try all usb port include hub nothing new, no saving power activate, all work great with the usb with another materiel.
I think my webcam have a bug. I will go on the store this saturday to change it

searching46dof, 2022-11-12:
@TheExium 

Asus latest chipset is  AMD Chipset driver V4.07.13.2243 For Windows 10/11 64-bit. 2022/09/30
https://rog.asus.com/us/motherboards/rog-strix/rog-strix-b550-f-gaming-model/helpdesk_download/

However AMD for b550 has 4.09.23.507 10/12/2022
https://www.amd.com/en/support/chipsets/amd-socket-am4/b550

In the release notes at https://www.amd.com/en/support/kb/release-notes/rn-ryzen-chipset-4-09-23-507
AMD USB4 CM Driver | 1.0.0.22 | Not Applicable | Added DFX support and few bug fixes
They don't indicate what the USB fixes are but might want to try the more recent chipset drivers.

searching46dof, 2022-11-12:
@Inhaltsangabe

VIDEO RESOLUTION | 1080p @ 30FPS / 720p @ 60FPS / 480p @ 30FPS / 360p @ 30FPS
https://www.razer.com/streaming-cameras/razer-kiyo/RZ19-02320100-R3U1#specs

Have you tried configuring the different resolutions and frame rates?
I would suggest from low to high starting at 360p @ 30FPS. 
360p = 480 x 360

lOXBetter, 2022-11-17:
may or may not help but ran into a similar issue with aitrack crashing. I had recently installed ndi tools and that created 4 ndi webcams. After uninstalling ndi 5 tools, the app started working again.

searching46dof, 2022-11-19:
try the instructions from the following
https://www.reddit.com/r/razer/comments/ktt7ww/razer_kiyo_fix_usb_video_device/

it seems like there may be an issue autodetecting the device type. there are other instances of this type of issue with other cameras as well.
https://www.reddit.com/r/Windows10/comments/igyt2s/webcams_camera_is_not_detected_but_audio_works/