AIRLegend/aitrack #172: Z axis value incorrect

Pieloth, 2022-09-11:
AItrack 0.7.0-alpha and Opentrack 2022.3.0, with Logitech C920 cam

**The Z value reported into Opentrack seems incorrect by a factor of 10**

When I stand at about 70cm from my webcam, Z is -6
When I go closer to the webcam, the Z value decrements towards 0
This does not happen for X and Y, when I move my head, X and Y seems to report correct values in cm

Maybe a missing multiplying factor for Z ?

Thanks for your help

Nicolas

AIRLegend, 2022-09-12:
Yeah, the scale is lower than before for the Z axis. However, this should be fixable via the curves within Opentrack. Simply change the Max range for the Z axis and you're should be good to go.

[image]

Pieloth, 2022-09-12:
Thanks for your reply
That's actually what I did to workaroud, but was wondering if this was intentional, to have lowered the Z axis scale, or is it something to fix?

BoomerGunship, 2022-09-14:
Love the better tracking accuracy in version 0.7.0!

Unfortunately, it seems I'm also facing this issue with the Z-axis. Previously on 0.6.6, the "Raw data tracker" on Opentrack starts at "-60" on the Z-axis, and moves up to ~"-40" when I get closer to the screen. 
But on 0.7.0, it starts at "-5", and only moves up to "-3" even when I get right up to my screen. Even setting the Max Input to 10cm doesn't really help, since it only detects 2 units in difference. This makes the Z-axis unusable, and now I could not look closely into the HUD or instruments when flying in DCS.

For now, I could workaround this by just manually zooming with the mouse's scroll wheel instead. Hoping there's a fix to this, since it's much easier and fun to look into the HUD/instruments through face tracking.

Cheers!

AIRLegend, 2022-09-14:
Just removed Z scaling (https://github.com/AIRLegend/aitrack/commit/63349832e0c64254fa1ac9fa77179558f93a8aed). Will be out with the next release.

Closing as fixed :)

closed 2022-09-14 by AIRLegend