AIRLegend/aitrack #183: AITrack crashes Windows

Omega-Projekt, 2022-09-17:
Hello
As soon as I am clicking on "Start tracking" the AITrack initiates a reboot of the PC.

I already did different changes in the settings of the Config (Resolution / FPS) but the App crashed every time.
I also replaced the installed VC++ 2015-2022 with the one form the Git but the PC is still rebooting.

[image]

**Environment (please complete the following information):**
 - Windows 10 Pro (21H2)
 - ASUS PRIME X299-DELUXE with i7 7820x
 - 16GB DDR4
 - RTX 2070 

I have checked the Event Viewer and there are no issues mentioned. Please see attached Log from the AITracker

[log.txt](https://github.com/AIRLegend/aitrack/files/9592023/log.txt)

Sorry that I can't provide more info but the PC just restarts without producing any more info that I could add.

With regards

searching46dof, 2022-09-20:
Rebooting the PC indicates a driver issue.
Does the video camera work with another application?
Is there any problem indicated in device manager?
Does the PC reboot unexpectedly under other circumstances?

You may want to verify the BIOS, chipset drivers and usb drivers are up to date.
Does Windows Event Viewer indicate any cause for the reboot? (e.g. exceptions)
You can use something like prime95 to check for the stability of the computer.
Also verify that the camera has sufficient power via the USB port. Try connecting via a powered usb hub.

An unusual item is that the log.txt indicates 60FPS while the screenshot indicates 30FPS.

powilliam, 2023-01-15:
Any thoughts on what's the cause and how to fix it? I'm also having this issue and Event Viewer logs as a power related issue aka `Kernel Power event id 41` but everything is normal with my power source and its readings on the motherboard bios

searching46dof, 2023-01-20:
@powilliam, a Kernel Power event id 41 indicates an unexpected shutdown.
verify that you have the latest chipset drivers, usb device drivers, and webcam drivers since a shutdown/reboot is usually driver related.
Is there a event viewer log entry immediately prior to the Kernel Power event ?
Have you run prime95 as a stress test for your cpu/memory and power? bios readings are usually when idle and you should check readings when under max stress/load to test for stability..
Most bios settings default to auto overclocked settings. You may want to check this and manually set them to fixed standard values.