AIRLegend/aitrack #184: Way too high of cpu usage.

muqtadadabest, 2022-09-18:
Whenever I run open track + aitrack and play a game, as soon as I start head tracking, it consistently stays at 15% cpu usage and sometimes peaking 20% cpu usage.

[image]   
These are regular settings and I haven't tweaked much at all. I can't half my resolution for my camera because it just crashes Ai Track.
Head tracking works fine, it's just it uses so much of my resources it makes my games almost unplayable and my computer starts to overheat.

searching46dof, 2022-09-20:
The CPU util seems unusually high. It's approximately double what I see for that resolution/frame rate on an i7-2600. What is your CPU/memory configuration? Generally lower camera resolution and framerate will reduce CPU.

You may want to check if the CPU and memory is running at the rated speed and also the CPU bus (also known as the Front Side Bus) as well. And also memory is running at it's rated speed.

Check that the BIOS and chipset drivers are up to date.

I also observe that it also depends on the make/model of the camera as well. Check that the camera's driver and usb driver are also up to date and the camera is running at USB2 or USB3 speeds.

Make sure that the camera has sufficient power and preferrably connected via a USB3 port. It may be under powered and running in low power mode.

You may also want to check your software configuration (e.g. antivirus) is not contributing the higher CPU.

zamzam666, 2022-11-06:
Im using a laptop camera so I'm not sure about usb speeds, my cpu is the i7-7700hq, it usually runs at around 3.60ghz, most of the time i lower it down to 3.39ghz. Bios is up to date, chipset drivers are also up to date.

searching46dof, 2022-11-07:
i've tested it on a i7-2600 desktop using a usb camera and an i7-2670 laptop with the built in camera which should be comparable. both configurations typically use approximately half what you are experiencing. is the laptop connected to an external power source?
if you select custom brightness and adjust gain and exposure for a normal exposure in the preview window, do you still have high cpu?
you may want to verify that memory and the cpu bus is running at their rated speeds and that you have sufficient memory to avoid swapping to disk.

searching46dof, 2022-11-07:
I was able to duplicate the high CPU reported by zamzam when there are very bright lights in the preview window background with insufficient lighting to the face.

zamzam, can you confirm there is enough direct lighting to your face for a normal exposure in the preview window?
I find aitrack works best with direct overhead lighting a little in front of the face.

zamzam666, 2022-11-08:
i'll try and fiddle with brightness settings and see what is going on.

zamzam666, 2022-11-08:
ok so brightness settings in aitrack, tried all types of combinations, didn't do much. Then I made a comparison with my light off and then light on, head tracking was worse with light off, however cpu speeds still remained at 15%.

zamzam666, 2022-11-08:
my laptop is always connected to my battery, also i have norton antivirus and i excluded the aitrack and opentrack folder so antivirus shouldn't be contributing anything.

searching46dof, 2022-11-09:
Can you provide your system configuration via exporting to a text file from msinfo32 ?
Reproducing the symptoms, can you provide screenshots of Windows Task Manager "processes" tab with Cpu Usage and all memory and I/O columns enabled/displayed and also a screenshot of the "peformance" tab ?

zamzam666, 2022-11-09:
so... i managed to get the cpu usage down to 9 - 10% by turning off my room light and using a ring light directly on my face. also idk how to do msinfo32 and stuff. But im ok with 9-10%, my pc is no longer overheating.