AIRLegend/aitrack #189: Proszę o pomoc

Pitrix690, 2022-09-27:
1. https://user-images.githubusercontent.com/114502368/192563023-41b4af8a-73fd-483c-9f47-301432eafb89.png
2. https://user-images.githubusercontent.com/114502368/192563026-6808cabe-9c64-416e-afe0-55b411159f59.png
3 .https://user-images.githubusercontent.com/114502368/192563009-35b1ae43-dbf1-4632-a9eb-c521d3b9ec37.png
4. https://user-images.githubusercontent.com/114502368/192563019-b1ddd1b3-2bae-47ed-97de-e409fd2ab3c1.png

W takich kolejnościach wyskoczyły komunikaty o błędach. Bardzo proszę o pomoc

searching46dof, 2022-10-08:
The screenshots seem to indicate missing dlls msvcp140.dll, vcomp140.dll, vcruntime140.dll and vcruntime140_1.dll

You can try the steps recommended at:
https://answers.microsoft.com/en-us/windows/forum/all/msvcp140dll-is-missing-in-my-win-10/1c65d6b0-68b8-4b59-b720-3e6a33038389
https://answers.microsoft.com/en-us/windows/forum/all/missing-vcomp140dll/23f750ee-99d4-4b34-99df-cf05a74457b7
https://answers.microsoft.com/en-us/windows/forum/all/vcruntime140dll/b1a15b0e-e389-41a3-bc49-f4fc85bac575

Pitrix690, 2022-10-08:
Witam, dziękuję bardzo za pomoc. Wszystko działa. Jeszcze raz dziękuję. Pozdrawiam