AIRLegend/aitrack #190: Crashing on startup/launch

Vertliz, 2022-10-05:
When I click the Ai Track.exe it launches the GUI and then it crashes. Please help!

searching46dof, 2022-10-08:
Please see https://github.com/AIRLegend/aitrack/issues/171 since it appears to be the same symptom

searching46dof, 2022-11-06:
I believe it saves the configuration to prefs.ini in the same folder so it would require write privilege.

I would prefer prefs.ini remain in the same folder so that I can easily run multiple instances concurrently. I'm working on an application to extend the functionality by averaging the udp stream data for better coverage and more accuracy through parallax.

Aberro, 2023-01-13:
Is it possible to at least have some logs to know what's wrong?

searching46dof, 2023-01-20:
crashes are recorded by windows in the windows event viewer logs.
Check that your login account has full rights to the folder.
Check windows settings that your account has privacy rights to the webcam.
And check if it may be blocked by antivirus. 

also see https://github.com/AIRLegend/aitrack/issues/183 which this may be a duplicate of.