AIRLegend/aitrack #193: vibrating blue square

scaevolapex, 2022-12-06:
Hi all,
I fly in DCS and I was very happy, when my friend told me about this project.
I downloaded and configurated everything, everything is working but not properly.
the blue square is vibrating, always and continously, which is a problem because I cant use a single switch in the cockpit.
I tried to fine tune both programs (AITrack & opentrack) tried to set deadzones but still not good.

Any idea before I buy the trackIR? :o)

AIRLegend, 2022-12-06:
Hey,

Try different models within AITrack, enable landmark stabilization and make sure you have good lightning.

searching46dof, 2022-12-25:
Try setting a deadzone of +/- 5deg from center when looking ahead to stabilize the front panel instruments. A chair headrest may also help. Also try enabling opentrack filters to reduce jitter when not in the deadzone.
If you want to use the headtracking explicitly for manipulating the cockpit switches, also try disabling x, y, z and roll mapping and only map yaw and pitch.

What scaling factor are you using in opentrack mapping properties ?
I would recommend 3x  (e.g. 30 deg maps to 90 deg) otherwise larger scaling factors create too much jitter and lower scaling factors do not have enough coverage.
see https://github.com/AIRLegend/aitrack/wiki/User-guide

As AIRLegend mentioned, good lighting is very important. Try overhead lighting on even during daytime.