AIRLegend/aitrack #199: Shattered Image, Low Fps

Furqan05, 2023-01-04:
I have a eken h9r action camera and i cant work it as my webcam at aitrack. In cam settings its 1920p 120fps,when i try to use aitrack at that setting this happens
[image]
And if i want to play at 720p it lags so much. Please Help.

searching46dof, 2023-01-10:
The manufacturer's web page indicates the H9R supports 4k30, 2.7k30, 1080p60,1080p30 and 720p200.

Using a higher resolution improves accuracy but uses more computing power and will depend on the rest of your system.
Using a lower resolution/frame rate will help improve responsiveness. I'm using 640x480@30fps with a PS3eye.

720p @ 200fps = 1280x720 * 200 =184,320,000
1080p @ 30fps = 1920x1080 * 30 =   62,208,000
try 1080p @30fps

Also verify that the camera is connected via usb3 to improve throughput and that it has sufficient power. Is the video rendered correctly with another application?
if the computer is not supplying sufficient power, you may need a powered usb3 hub.

Emanolinis, 2023-07-09:
> The manufacturer's web page indicates the H9R supports 4k30, 2.7k30, 1080p60,1080p30 and 720p200.
>
> Using a higher resolution improves accuracy but uses more computing power and will depend on the [...]

Thanks this solved everything!