AIRLegend/aitrack #202: app doesnt respond when opened

Lacakali-hu, 2023-01-09:
**Describe the bug**
when i open ai track it opens but doesnt respond

**To Reproduce**
Steps to reproduce the behavior:
1. open ai track
2. ai track doesnt respond
https://user-images.githubusercontent.com/106344453/211303224-475ee2a9-4dd9-4e40-b721-9d459eec7c64.png

**Expected behavior**
ai track should open and work noramaly

**Screenshots**
https://user-images.githubusercontent.com/106344453/211302745-b2a8c302-af33-422e-9090-7d0fa7b00bc5.png
https://user-images.githubusercontent.com/106344453/211303224-475ee2a9-4dd9-4e40-b721-9d459eec7c64.png

**Environment (please complete the following information):**
 - Operating system win 11 22h2
 - Webcam model: 
 - AITrack version: 0.6.4 0.7.0 0.7.1

searching46dof, 2023-01-10:
please verify that:
- your user account has rights to the folder.
- that aitrack isn't blocked by antivirus.
- that windows settings/privacy/camera allows aitrack to access the camera.

I'm running Windows 10. Do you have a WIndows 10 installation for testing?

Lacakali-hu, 2023-01-10:
for some reason now it works whitout a problem but thanks for the help

closed 2023-01-10 by Lacakali-hu