AIRLegend/aitrack #204: FOV, DISTANCE and RESOLUTION changed. Now my face isn't recognized anymore, unless I move really close to the camera

anakinsolo, 2023-01-15:
**Describe the bug**
FOV, DISTANCE and RESOLUTION changed. Now my face isn't recognized anymore, unless I move really close to the camera

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "Configuration"
2. Changed RESOLUTION to 1280 width and 720 height (width and height were lower)
3. Changed FOV to 78 (it was 60)
4. Calibrate

**Expected behavior**
Face isn't recognized anymore. When face moves, like side way, it detects a little. When face is looking straight forward, there isn't the blue square and pink dots anymore

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Operating system version (Windows 10+ Version number): Windows 10 latest build
 - Webcam model: C922 Pro
 - AITrack version: 0.7.1

**Additional context**
It had been working really nicely until I decided to read through the wiki and changed the values. There isn't a way to reset to old values or backup so I can't go back.

searching46dof, 2023-01-20:
if you stop tracking, you should be able to click the configure button and re-edit the fields. and click apply to save the settings.
otherwise you can re-extract the downloaded archive to reset aitrack.

according to the logitech C922 spec's you should be able to use 1280x720 and 78 for FOV.
the C922 Pro's autozoom may be causing an issue with the headtracking since it interferes with most head tracking software. you may want to try manually disabling it. it can account for the lower resolution and FOV.

anakinsolo, 2023-02-12:
> if you stop tracking, you should be able to click the configure button and re-edit the fields. and click apply to save the settings. otherwise you can re-extract the downloaded archive to reset [...]

hello, thanks for replying. Autozoom disabling does not help.
The best res for tracking is 640 x 480, 60 fps. Distance is at 0.7m and FOV is 72.

But it keeps losing track during a session. For example, after 30 mins, it loses the face detection even though my face is at a stand still.

searching46dof, 2023-02-17:
what type of lighting are you using? direct overhead lighting works best. there may be some issues with detection using side lighting or insufficient ambient lighting. backlighting will cause problems. try making the room brighter. the lighting is the most common cause for losing tracking.

if you are using this for a flight simulator, I would recommend placing the webcam above the monitor for dogfighting so that it is more sensitive and has more detection range when looking above rather than below.

are you using aitrack with a desktop monitor size or with a tv at a distance? version 0.70 was a little too aggressive at optimization shrinking the facial detection bounding box. version 0.71 increases the bounding box but it still limits the detection to approximately 2 meters away dependent on a specific face.  (see issue https://github.com/AIRLegend/aitrack/issues/173) I've proposed a solution that the smallest bounding box should be a percentage of the webcam's width and height resolution which appears to work at distances of 4 meters. 

does it also lose tracking for a static face such as a 8x10 portrait or face on a magazine? if it does lose tracking, there may be a performance issue involved. e.g. cpu, memory, chipset drivers, usb.

anakinsolo, 2023-03-10:
> what type of lighting are you using? direct overhead lighting works best. there may be some issues with detection using side lighting or insufficient ambient lighting. backlighting will cause [...]

hi, thanks for the reply. It is usually overhead direct light. But sometimes I have lights coming from behind because of my wife's monitors, it makes the camera loses the tracking. But nonetheless, I just need to block the lights from my back and the tracking should be ok

> if you are using this for a flight simulator, I would recommend placing the webcam above the monitor for dogfighting so that it is more sensitive and has more detection range when looking above [...]

MSFS 2020, Falcon 4 BMS and War Thunder are the only games. And yes, I mount the webcam on the top side of my monitor, I think it's a 28" one. I have always mounted it that way, even if trackIR back then 3 4 years ago.

> does it also lose tracking for a static face such as a 8x10 portrait or face on a magazine? if it does lose tracking, there may be a performance issue involved. e.g. cpu, memory, chipset drivers, usb.
No, not really. And I do not think it's performance issue. But for static face, it has this weird behavior that the inputs are different between ingame input and raw input. Meaning that, you stay absolutely still. Raw input should be 0. But ingame input, there are some minor changes, which makes it impossible to stay put in game, there is always some drift