AIRLegend/aitrack #207: Can't find the .exe file

czross, 2023-02-13:
**Describe the bug**
I might be having a dumb moment, but I cannot for the life of me find where the .exe file is either in the GitHub repo or on my personal copy. 

**To Reproduce**
download files and try and find the dang .exe file.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

**Environment (please complete the following information):**
 - Operating system version (Windows 10+ Version number): 
 - Webcam model: 
 - AITrack version:

**Additional context**
Add any other context about the problem here.

searching46dof, 2023-02-17:
on the main webpage, click on releases on the right side.
then click on a specific release e.g. 0.71 which will provide a link for downloading that release archive

czross, 2023-02-17:
Yup just me being a goober, thanks!

closed 2023-02-17 by czross