AIRLegend/aitrack #222: Camera preview is off center at half point.

gdomjan83, 2023-04-29:
**Describe the bug**
When I launch the preview, although I sit in front of the webcam, the preview is showing like in the attached picture. It is off center. This causes the tracker to lose track of my face.

**To Reproduce**
Just launching the application.

 - Operating system version (Windows 10+ Version number): Windows 10 22H2
 - Webcam model: Logitech C600
 - AITrack version: v.07.1

[image]

gdomjan83, 2023-04-29:
Update: It turned out that the problem was the camera. I tried with another camera, and it worked perfectly, but this Logitech C600 is not working correctly with the software.

AIRLegend, 2023-04-30:
This tends to happen when you use wrong FPS/Resolution config parameters. Make they are supported by your camera.

Aruany, 2023-06-11:
I have the same problem, once I change the resolution, the software doesn't work

Aruany, 2023-06-11:
My camera resolution is supposed to be 4K, but as soon as I enter that resolution, and start previewing, the software crashes