AIRLegend/aitrack #224: Linux Build

HealingArcYT, 2023-05-02:
Can you maybe release a Build for Linux OSes?

ProjectSynchro, 2023-09-09:
This actually seems pretty feasible as there isn't really any references to windows except for what's in 

[Client/src/utils/qglobalshortcuts/qglobalshortcut.cc](https://github.com/AIRLegend/aitrack/blob/ed7d58c7decbe47c12c1008d44ea00fa9fb28c87/Client/src/utils/qglobalshortcuts/qglobalshortcut.cc#L10)

As the GUI is QT that should work with some build system tinkering, the other libraries appear to have linux releases available as well.. (ONNX, lubusb, OpenCV) so this project may only be a few patches away from building on Linux, I'll have to try this out when I get a chance.

zimri-leisher, 2023-09-10:
Hey @ProjectSynchro , let me know if you end up working on this. Would love to spend some time to get this working.

tuxtergames, 2023-09-21:
@ProjectSynchro any news about linux version of this software? is it possivel to run it trought wine?

ProjectSynchro, 2023-09-21:
> @ProjectSynchro any news about linux version of this software? is it possivel to run it trought wine?

May have some time coming up this weekend to work on this, work has been killing me. 

The windows version works though wine. Just pair it with a Linux build of OpenTrack. You'll want to make sure you build OpenTrack with Wine support enabled.

Also expect a bit of CPU cost with the software. Still seems to work pretty well.

tuxtergames, 2023-09-21:
> May have some time coming up this weekend to work on this, work has been killing me.
>
> The windows version works though wine. Just pair it with a Linux build of OpenTrack. You'll want to make sure [...]

I can run it here, tks, now my problem is to make it work with opentrack, tks mate!!!!

MichaelTunnell, 2024-05-29:
Would it be a viable usecase for this project to use this in conjunction with some kind of desktop tools for turning this into an accessibility feature for those with limited mobility? If yes, then I would like to request native Linux support be added in addition to WINE because the value this could offer people would be enormous.

DeltaTimo, 2024-10-17:
I'm a bit late to the party, but it stopped working with wine on my machine, I got fed up and decided to spend today working on a port and I'm happy to say it paid off: https://github.com/DeltaTimo/aitrack