AIRLegend/aitrack #240: Camera's not found

XeXxes, 2023-08-22:
When I open AiTrack it says that no cameras were found. and in the configure menu there's no cameras to select. My webcam works fine with Skype, camera app, and discord. But no AiTrack. I've asked over 5 people in DCS discord for help and no one has a solution. It says check models folder but I don't know what to look for.
[image]
[image]
[image]
[image]
[image]

Rycia, 2023-10-26:
Also have this issue.

MrMellow01, 2024-04-18:
Yeah I've got this issue too any fix yet?

Mayyyk, 2024-09-23:
Did you fix it?