AIRLegend/aitrack #243: Aitrack doesn't detect my kinect

Kiaomidizadeh, 2023-09-14:
I have recently installed aitrack for sim racing head tracking. I plugged in my xbox 360 kinect to pc. And i opened aitrack. And i faced this message: "no cameras detected. Plug one and restart the program". Then when i click ok, another error shows up: there was a problem initializing the tracker. Check the models folder and restart the program. 
I wanted to know how can i solve this issue.
P.s: i also have the opentrack installed.
I have to mention that i have installed all kinect drivers (sdk and dev toolkit) according to the instructions
I tried to delete all kinect apps first, then i installed them in order. And then i plugged in the kinect. And still aitrack doesn't work

cdtatoo, 2023-11-08:
hi, I have a Kinect 2.0 and when start capture program crash. Solution? or is it incompatible?

SebastianAtWork, 2023-11-12:
I have this Problem too

cdtatoo, 2023-12-13:
solved. when you choose webcam in aitrack it's automatically set to 640*480 but kinetic have 1920*1080 an program crash. set right resolution and and 60 hz, aitrak work.