AIRLegend/aitrack #255: AITrack.exe not opening

Mallers007, 2024-02-25:
Very sorry to add another similar bug!  Probably something i am doing, but I have downloaded and unzipped the AITrack files, and I can see AIrack.exe as an application.  I double click to run it, and nothing happens.  Nothing briefly happens, i get no error message, i get no windows cursor going around which usually means it doing something.  I go to Task Manager and there's no AITrack there either.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'File Manager'
2. Click on 'AITrack.exe'
3. 3. Nothing happens

**Expected behavior**
Unsure, i guess program starts?

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Operating system version (Windows 10+ Version number): Win 11 Pro version 10.0.22621 build 22621 
 - Webcam model: Logitech C922
 - AITrack version: v0.7.0

**Additional context**
Add any other context about the problem here.

BoodooM, 2024-03-02:
Same issue here.
Tried to unzip with windows to C:\, system asked for administration rights, AITrack.exe don't start.

Solved:

Used 7Zip to unzip it.
Unziped it to another drive/folder

Mallers007, 2024-03-02:
> Same issue here. Tried to unzip with windows to C:, system asked for administration rights, AITrack.exe don't start.
> 
> Solved:
> 
> Used 7Zip to unzip it. Unziped it to another drive/folder

i ended up waiting for the next version to be released, downloaded that and works ok now!