AIRLegend/aitrack #258: Image is not showing, blinking dark screen or vertical lines

eBrosovitzki, 2024-03-05:
I've tried with my notebook built-in camera and a USB camera.  Version v0.7.1.
This is what I get when I start tracking:

[image]
[image]

Here's log file:

[2024-03-05 11:27:29.877] [aitrack] [info]  ----------  AITRACK LOG   ----------
[2024-03-05 11:27:29.878] [aitrack] [info] Created/Found prefs.ini
[2024-03-05 11:27:30.294] [aitrack] [info] Searching for cameras...
[2024-03-05 11:27:31.974] [aitrack] [info] Number of recognized cameras: 1
[2024-03-05 11:27:31.976] [aitrack] [info] UDP sender reinitialized. IP: 192.168.10.15  PORT: 4242
[2024-03-05 11:27:31.976] [aitrack] [info] Building Tracker with selected camera: 0
[2024-03-05 11:27:32.009] [aitrack] [info] Tracker initialized.
[2024-03-05 11:27:32.009] [aitrack] [info] Updated stabilizer.
[2024-03-05 11:27:32.009] [aitrack] [info] Updating camera parameters...
[2024-03-05 11:27:32.009] [aitrack] [info] Updated camera parameters. 640x360@30
[2024-03-05 11:27:32.010] [aitrack] [info] Checking for updates
[2024-03-05 11:27:32.049] [aitrack] [info] App initialized
[2024-03-05 11:27:46.283] [aitrack] [info] Saving prefs
[2024-03-05 11:27:46.283] [aitrack] [info] Updated stabilizer.
[2024-03-05 11:27:46.283] [aitrack] [info] UDP sender reinitialized. IP: 127.0.0.1  PORT: 4242
[2024-03-05 11:27:46.283] [aitrack] [info] Updating camera parameters...
[2024-03-05 11:27:46.283] [aitrack] [info] Updated camera parameters. 640x360@30
[2024-03-05 11:27:46.283] [aitrack] [info] Tracker initialized.
[2024-03-05 11:27:46.283] [aitrack] [info] Prefs saved
[2024-03-05 11:27:48.898] [aitrack] [info] Starting camera 0 capture
[2024-03-05 11:27:50.372] [aitrack] [info] Camera 0 started capturing
[2024-03-05 11:28:02.707] [aitrack] [info] Stop camera 0 capture

GigaHugo, 2024-04-04:
A reasonably reliable workaround for this problem is to restart the computer and start AItrack as soon as possible.

BlueFalcon069, 2024-04-06:
Hmm, still more or less of a hack though, I heard here that uninstalling 3rd party antivirus and doing a registry edit might fix: https://github.com/opentrack/opentrack/issues/604