AIRLegend/aitrack #261: Not working with virtual camera app (android)

Diordi7, 2024-03-26:
**Describe the bug**
I use my phone as my webcam, using an app called DroidCam. When I try to start AITrack with this cam selected, AITrack just crashes.

**To Reproduce**
Steps to reproduce the behavior:
1. Using an Android device, download DroidCam from play store, and the client on PC, and connect the cam using it.
2. Start AITrack, configure the camera to use DroidCam input, and try to start tracking.

**Expected behavior**
DroidCam is supposed to be a wireless input, so I just expected AIRTrack to accept it as any other cam input.

**Screenshots**
N/A. App just CTD.

**Environment (please complete the following information):**
 - Operating system version (Windows 10 Pro Version 10.0.19045 Build 19045): 
 - Webcam model: Android Samsung M51 - DroidCam app
 - AITrack version: v0.7.1

**Additional context**
I use my phone as a cam via DroidCam quite often, and it usually behaves like a regular webcam, so I have no idea why it`s not working here.

Ousadanken, 2024-04-28:
DEAD

Mayyyk, 2024-09-23:
did you fix it?

De-Clan-C, 2024-10-08:
This just happened to me. Here's the output I got. I'm using wine to run it on linux, but I'm assuming the out put would be the same. 
```

wine: Unhandled page fault on read access to FFFFFFFF6C8F0010 at address 00006FFFFFC5933D (thread 011c), starting debugger...
0168:fixme:dbghelp:elf_search_auxv can't find symbol in module
0168:fixme:dbghelp:elf_search_auxv can't find symbol in module

```