AIRLegend/aitrack #263: Does not work with star citizen

StrandedTimelord, 2024-04-05:
**Describe the bug**
The game does not detect the signal from the computer. I have enabled track IR within the game, but nothing is working.

**To Reproduce**
log into star citizen
turn on head tracking software,
the camera does not move 

**Expected behavior**
the camera should move along with IRL head.

**Environment (please complete the following information):**
 - Operating system version (Windows 10+ Version number): Windows 11 home, 23h2
 - Webcam model: Logitech c922 pro
 - AITrack version:0.7.1 alpha

nickpolet, 2024-04-20:
I reckon to need to be using [OpenTrack](https://github.com/opentrack/opentrack) alongside this as well for it to work.

I've had it working with Star Citizen for quite some time now.

artyPD, 2024-04-23:
@StrandedTimelord
Just to renforce what @nickpolet is saying. Pairing it with [OpenTrack](https://github.com/opentrack/opentrack) does work with most titles. Even obscure ones like [Tiny Combat Arena](https://store.steampowered.com/app/1347550/Tiny_Combat_Arena/) and [IL2](https://store.steampowered.com/app/307960/IL2_Sturmovik_Battle_of_Stalingrad/). Works like a charm :)

StrandedTimelord, 2024-04-26:
I have tried everything. I did use opentrack with ai track. No such luck. Im closing this thread as nothing is working, and im moving on.

closed 2024-04-26 by StrandedTimelord