AIRLegend/aitrack #267: AItrack didn't detect the Droidcam

ngmingzhe, 2024-07-02:
I use Droidcam as my computer camera but the AItrack didnt detect my camera.
[image]
The Droidcam shows in "Sound,video and games controller" which i think the AItrack only can detect the "Camera".
What is the solution to solve the problem in order to let the AItrack detect droidcam?

Mayyyk, 2024-09-23:
Did you manage to fix it?

Gerbooo, 2025-03-14:
Hello,

I had the same problem the only reasonable fix I could find is to **revert the version back to _V0.6.5_ _Alpha_**
If you find any other fixes pls let me know bc I also want the latest version

efepdrk, 2025-08-03:
Hey, have you or anyone else found a fix for this?