AIRLegend/aitrack #279: AI Track does not find my Webcam

Tobinal95, 2024-12-21:
Hi, I have been using Ai Track for a while and now I have the problem that the application cannot find my webcam and therefore the app does not start.

[image]

Thank you in advance

Gerbooo, 2025-03-14:
Hello,

I had the same problem the only reasonable fix I could find is to **revert the version back to _V0.6.5_ _Alpha_**
If you find any other fixes pls let me know bc I also want the latest version

nmanuelides, 2025-10-02:
Same here, and this has been an issue for years apparently but no one bothers to fix it. Tried with just my webcam (genius E8000), with camo studio, with obs, with Iriun using my android phone, nothing works. The weirdest thing is that the first time I tried with camo it worked, then I uninstalled camo because I did not have the paid version and the output looked like shit and after that it never worked again, even after reinstalling camo